uPesy/easyeda2kicad.py #1: Save as .kicad_sym (instead of legacy format) & add more details In symbol?

glowingkitty, 2022-01-25:
Is there a way to directly write a .kicad_sym file, so it can be edited in KiCad 6?
Also, would be super useful to have more details from the web page available in the symbols - LCSC ID, description, datasheet, package, and price per unit are the most important which I have always added manually so far.

uPesy, 2022-02-14:
I have not switch to KiCad 6 so i don't really know.
It's a great idea to have the price per unit and also the availability.
But i don't know how to dynamically get the current stock and price of components. 
Maybe with a KiCad plugin

glowingkitty, 2022-02-14:
https://github.com/Bouni/kicad-jlcpcb-tools Has those data

uPesy, 2022-02-15:
Thanks for the share !
This plugin seems really great. 
But i don't know if a kicad plugin can modify a symbol library.

Bouni, 2022-05-03:
I would like to reopen this one.

Adopting the new symbol format (.kicad_sym) would be great because its is not possible to edit symbols in the legacy format.
As easyeda2kicad sometimes generates symbols with minor issues I would love to edit these and fix by hand which is still a 1000 times faster that creating the symbol by hand. KiCAD allows to migrate the from .lib to .kicad_sym but that means that easyeda2kicad cannot add new symbols anymore.

Do you still want to maintain V5 and V6? In my opinion you should go ahead and support V6 + as its the official version for quite a while now.

uPesy, 2022-05-19:
You can now save in .kicad_sym for Kicad v6 in v0.3.
It will save by default in .kicad_sym format.
You can still use the legacy format using the `--v5` argument
#28

closed 2022-05-19 by uPesy