uPesy/easyeda2kicad.py #2: Conversion errors.

pierreSCpapon, 2022-02-14:
First off, thank you for this amazing tool, it really speeds up development whilst using LCSC as part supplier.

I wanted to signal some import errors I have noticed:

1) Rotation of pad is wrong on converted footprint
Part in question:
https://lcsc.com/product-detail/Low-Dropout-Regulators-LDO_UMW-Youtai-Semiconductor-Co-Ltd-XC6206P332PR_C545581.html

EasyEDA preview:
[image]

Converted footprint:
[image]

2) Symbol shape missing (seems to be triangles)
Part in question:
https://lcsc.com/product-detail/Schottky-Barrier-Diodes-SBD_DIODES_1N5819HW-7-F_1N5819HW-7-F_C82544.html

EasyEDA preview:
[image]

Converted symbol:
[image]

more examples:
https://lcsc.com/product-detail/Light-Emitting-Diodes-LED_FM-B2020RGBA-HG_C108793.html
[image]
[image]

https://lcsc.com/product-detail/Others_HYP-1TS009C-2600-5000_C319427.html
[image]
[image]

Let me know if you need more info or would like me to test anything, thank you.

uPesy, 2022-02-14:
@pierreSCpapon Thanks for your report.
I have also these issues when testing.
For the footprint, there is a rotation and an offset issue. Do you have notice this behaviour with others components when there is an uncommon pad ?
For the symbol, i have noticed that EasyEda add 2 more types : Polygon and path.
You should have warning when you run the script right ?
```
[*] Creating Kicad symbol library for LCSC id : C108793
        [-] Unknow symbol designator : PG
        [-] Unknow symbol designator : PG
        [-] Unknow symbol designator : PG
        [-] Unknow symbol designator : PG
        [-] Unknow symbol designator : PG
        [-] Unknow symbol designator : PG
        [-] Unknow symbol designator : PG
        [-] Unknow symbol designator : PG
        [-] Unknow symbol designator : PG
[*] Creating Kicad footprint library for LCSC id : C108793
```
I will try to implement that this week-end.
Thanks you again to point out issue on some components since i can't test them all :)

uPesy, 2022-02-15:
I have fix the issues for symbol.
Feel free to report other bugs. I will fix the footprint issue later.

pierreSCpapon, 2022-02-17:
Awesome! Thank you.

I have actually just tested the new version and found an offset error with the symbol.

https://lcsc.com/product-detail/Buzzers_S-S-SFN-17075PA10_C360608.html

uPesy, 2022-03-06:
All the symbols issues mentionned previously should be fixed now in the new release #18 
You can update the symbol by using `--overwrite` argument 
```
easyeda2kicad --symbol --footprint --lcsc_id=C319427  --overwrite
```

closed 2022-04-13 by github-actions[bot]