uPesy/easyeda2kicad.py #20: Offset issue for 3D Model

uPesy, 2022-03-19:
Hi,
I have not find the way to convert the offset of easyeda 3d model to KiCad 5.x

For exemple for ID C319148, here the JSON with only relevant data
```json
"dataStr": {
                "head": {
                    "x": 4001.992,    // x origin footprint
                    "y": 2999.8101,  // y origin footprint
                },
                "canvas": "CA~1000~1000~#000000~yes~#FFFFFF~10~1000~1000~line~0.03937~mm~1~45~visible~0.5~4001.992~2999.8101~0~none",
                "shape": [
                    "SVGNODE~{\"gId\":\"g1_outline\",\"nodeName\":\"g\",\"nodeType\":1,\"layerid\":\"19\",\"attrs\":{\"c_width\":\"35.19678\",\"c_height\":\"31.1023\",\"c_rotation\":\"0,0,180\",\"z\":\"-3.5433\",\"c_origin\":\"4001.992,3004.2501\",\"uuid\":\"c50ce21ae3b941ce8eeb9bccc6cdef41\",\"c_etype\":\"outline3D\",\"id\":\"g1_outline\",\"title\":\"USB-3.1-SMD_16P-L8.9-W7.3-H3.3-P0.5\",\"layerid\":\"19\",\"transform\":\"scale(1) translate(0, 0)\"},\"childNodes\":[{\"gId\":\"g1_outline_line0\",\"nodeName\":\"polyline\",\"nodeType\":1,\"attrs\":{\"fill\":\"none\",\"id\":\"g1_outline_line0\",\"c_shapetype\":\"line\",\"points\":\""}}]}"
                ],
```

uPesy, 2022-06-27:
I don't know how to properly set the offset of all 3D model for x, y , z offset. Here some relevant information : 
For example for the 3D model of LCSC ID C319148
API result : [https://easyeda.com/api/products/C319148/components?version=6.4.19.5](https://easyeda.com/api/products/C319148/components?version=6.4.19.5)
Here the data for the 3D model : 
```json
{
   "gId":"g1_outline",
   "nodeName":"g",
   "nodeType":1,
   "layerid":"19",
   "attrs":{
      "c_width":"35.19678",
      "c_height":"31.1023",
      "c_rotation":"0,0,180",
      "z":"-3.5433",
      "c_origin":"4001.992,3004.2501",
      "uuid":"c50ce21ae3b941ce8eeb9bccc6cdef41",
      "c_etype":"outline3D",
      "id":"g1_outline",
      "title":"USB-3.1-SMD_16P-L8.9-W7.3-H3.3-P0.5",
      "layerid":"19",
      "transform":"scale(1) translate(0, 0)"
   },
   "childNodes":[
      {
         "gId":"g1_outline_line0",
         "nodeName":"polyline",
         "nodeType":1,
         "attrs":{
            "fill":"none",
            "id":"g1_outline_line0",
            "c_shapetype":"line",
            "points":"4019.5904 3019.7618 4019.5904 3019.7224 4019.5904 3019.6831 4019.5904 3012.8327 4019.5904 3012.7933 4019.5904 3012.7539 4019.5904 3012.7146 4019.5904 3012.6752 4019.5904 3012.5571 4019.5904 3012.5177 4019.5904 3012.3996 4019.5904 3012.3602 4019.5904 3012.1634 4019.5904 3011.9272 4019.5904 3011.691 4019.5904 3011.4941 4019.5904 3011.3366 4019.5904 3011.2973 4019.5904 3011.1791 4019.5904 3011.1398 4019.5904 3011.1004 4019.5904 3011.061 4019.5904 3011.0217 4019.5904 3010.9823 4019.5904 3010.8642 4019.5904 3010.8248 4019.5904 3010.628 4019.5904 3010.5886 4019.5904 3010.313 4019.5904 3009.9587 4019.5904 3009.565 4019.5904 3009.1713 4019.5904 3008.8563 4019.5904 3008.8169 4019.5904 3008.5414 4019.5904 3008.502 4019.5904 3008.3051 4019.5904 3008.2658 4019.5904 3008.1477 4019.5904 3008.1083 4019.5904 3008.0689 4019.5904 3008.0295 4019.5904 3007.9902 4019.5904 3007.9508 4019.5904 3007.8721 4019.5904 3007.8327 4019.5904 3007.754 4019.5904 3005.9429 4019.5904 3005.9036 4019.5904 3005.8642 4019.5904 3005.7855 4019.5904 3005.7461 4019.5904 3005.7067 4019.5904 3005.6673 4019.5904 3005.628 4019.5904 3005.5886 4019.5904 3005.4705 4019.5904 3005.4311 4019.5904 3005.313 4019.5904 3005.2736 4019.5904 3005.0768 4019.5904 3004.8406 4019.5904 3004.6044 4019.5904 3004.4075 4019.5904 3004.25 4019.5904 3004.2107 4019.5904 3004.0925 4019.5904 3004.0138 4019.5904 3003.9744 4019.5904 3003.9351 4019.5904 2997.0847 4019.5904 2997.0059 4019.5904 2996.9272 4019.5904 2996.8485 4019.5904 2996.7697 4019.5904 2996.691 4019.5904 2996.6122 4019.5904 2996.5335 4019.5904 2996.4154 4019.5904 2996.1792 4019.5904 2995.9823 4019.5904 2995.7067 4019.5904 2995.5493 4019.5904 2995.3524 4019.5904 2995.2343 4019.5904 2995.1162 4019.5904 2995.0374 4019.5904 2994.88 4019.5904 2994.6831 4019.5904 2994.6044 4019.5904 2994.3682 4019.5904 2994.2894 4019.5904 2994.0138 4019.5904 2993.6201 4019.5904 2992.754 4019.5904 2992.6359 4019.5904 2992.2422 4019.5904 2991.8878 4019.5904 2991.5729 4019.5904 2991.3367 4019.5904 2991.1792 4019.5904 2991.1398 4019.551 2991.1004 4019.4723 2991.0611 4019.4329 2991.0611 4019.3148 2991.0611 4019.2361 2991.0611 4018.9605 2991.0611 4018.5668 2991.0611 4018.4881 2991.0611 4018.4093 2991.0611 4018.3306 2991.0611 4018.2518 2991.0611 4018.2125 2991.0611 4018.0944 2991.0611 4018.0156 2991.0611 4017.9762 2991.0611 4017.8188 2991.0611 4017.7794 2991.0611 4017.6613 2991.0611 4017.5432 2991.0611 4017.3857 2991.0611 4017.307 2991.0611 4017.2282 2991.0611 4017.1101 2991.0611 4016.9526 2991.0611 4016.8345 2991.0611 4016.7164 2991.0611 4016.5589 2991.0611 4016.3227 2991.0611 4016.2833 2991.0611 4016.2046 2991.0611 4015.6534 2991.0611 4015.4566 2991.0611 4015.4566 2988.6989 4013.8818 2988.6989 4013.8818 2991.0611 4012.4251 2991.0611 4012.4251 2988.6989 4010.8503 2988.6989 4010.8503 2991.0611 4010.4959 2991.0611 4010.4566 2991.0611 4009.9448 2991.0611 4009.7479 2991.0611 4009.6692 2991.0611 4009.2755 2991.0611 4009.2755 2988.6989 4008.4881 2988.6989 4008.4881 2991.0611 4008.0156 2991.0611 4007.9763 2991.0611 4007.7794 2991.0611 4007.74 2991.0611 4007.5038 2991.0611 4007.4645 2991.0611 4007.307 2991.0611 4007.307 2988.6989 4006.5196 2988.6989 4006.5196 2991.0611 4005.3385 2991.0611 4005.3385 2988.6989 4004.5511 2988.6989 4004.5511 2991.0611 4003.37 2991.0611 4003.37 2988.6989 4002.5826 2988.6989 4002.5826 2991.0611 4001.4015 2991.0611 4001.4015 2988.6989 4000.6141 2988.6989 4000.6141 2991.0611 3999.433 2991.0611 3999.433 2988.6989 3998.6456 2988.6989 3998.6456 2991.0611 3997.4645 2991.0611 3997.4645 2988.6989 3996.6771 2988.6989 3996.6771 2991.0611 3996.5196 2991.0611 3996.4802 2991.0611 3996.244 2991.0611 3996.2046 2991.0611 3996.0078 2991.0611 3995.9684 2991.0611 3995.496 2991.0611 3995.496 2988.6989 3994.7086 2988.6989 3994.7086 2991.0611 3994.3149 2991.0611 3994.2755 2991.0611 3993.5275 2991.0611 3993.4881 2991.0611 3993.1338 2991.0611 3993.1338 2988.6989 3991.559 2988.6989 3991.559 2991.0611 3990.0629 2991.0611 3990.0629 2988.6989 3988.4881 2988.6989 3988.4881 2991.0611 3988.3306 2991.0611 3987.7795 2991.0611 3987.7007 2991.0611 3987.6613 2991.0611 3987.2676 2991.0611 3987.1495 2991.0611 3987.1102 2991.0611 3986.9921 2991.0611 3986.7558 2991.0611 3986.6377 2991.0611 3986.5984 2991.0611 3986.4015 2991.0611 3986.3228 2991.0611 3986.2047 2991.0611 3986.1653 2991.0611 3985.9684 2991.0611 3985.8897 2991.0611 3985.8503 2991.0611 3985.6928 2991.0611 3985.6535 2991.0611 3985.5354  [... cut ...]
https://github.com/uPesy/easyeda2kicad.py/blob/6325431f44c1338f1f3b72f0f8da8dacb161b2e1/easyeda2kicad/kicad/export_kicad_footprint.py#L204-L211
Some kind of formula should be found between `"c_origin"`, `"x"` and `"y"` footprint bbox, `c_width` and `c_height` , and maybe with the other `"BBox"` key too.

I try to look the code on EasyEDA, it's React production code, to find the formula. I found this, but i have not managed to get relevant offset : 
In line 87700 of `main.min.js` : 
[image]

Your help is really appreciated:)

Note that in the end, the dimensions should be converted in mm for Kicad

Bouni, 2022-06-27:
I'll take a look, probably this week, very busy at the moment ....

closed 2022-09-10 by github-actions[bot]