uPesy/easyeda2kicad.py #22: ModuleNotFoundError: No module named 'requests'

SnowballsChance, 2022-04-18:
Hi,

I am new to using python and suspect I have a configuration error but would really appreciate some advice please.

Environment: 
macOS: 12.2.1
M1 Macbook Air
Python: 3.9.12
Installed via Brew

Have uninstalled and reinstalled easyeda2kicad but get same error.

```
snowballsChance @192.168.112.223 ~ % easyeda2kicad --full --lcsc_id=C2040
Traceback (most recent call last):
  File "/opt/homebrew/bin/easyeda2kicad", line 5, in <module>
    from easyeda2kicad.__main__ import main
  File "/opt/homebrew/lib/python3.9/site-packages/easyeda2kicad/__main__.py", line 8, in <module>
    from easyeda2kicad.easyeda.easyeda_api import easyeda_api
  File "/opt/homebrew/lib/python3.9/site-packages/easyeda2kicad/easyeda/easyeda_api.py", line 2, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
```

I appreciate any thoughts you might have.

SnowballsChance, 2022-04-18:
aah - Actually I fixed it.

`pip install requests`

it runs now!
sorry for the distraction

closed 2022-04-18 by SnowballsChance

uPesy, 2022-04-18:
Thanks for pointing out this issue. In fact, i forget to add the lib `requests` in the project dependencies. 
Fixed in #23