uPesy/easyeda2kicad.py #26: Feature request: Add GUI

Bouni, 2022-04-26:
I love this plugin but it feels not really integrated. I think a GUI woud be absolutely awesome!

It could be super simple, maybe just a single window that shows up after the user clicks the action plugin button.
There you got the options (checkboxes, radios, whatever) and paths which could be saved in a YAML/TOMS/INI file beside the plugin itself.
And a textfield for the LCSC number plus a Go button to download the data.

Actually not that big of a deal to build and would make the plugin usable from within KiCAD.

uPesy, 2022-04-30:
It's a great idea that i have also in mind.
But i don't know how actually do it on Kicad. The plugins should be compatible for v5+ et v6.
I know that it's more easier to integrate plugins in Kicad 6. A GUI similar to yours https://github.com/bouni/kicad-jlcpcb-tools will be really great.
You can submit a PR with template code for a GUI inspired from kicad-jlcpcb-tools.

Bouni, 2022-05-03:
I thought a bit more about this and in the end haveing the GUI just out of the PCB editor is kind of limited.
Maybe a standalone GUI would be a better choice (also not a big deal).
If I find some spare time I'll submit a PR with a GUI approach ...

uPesy, 2022-05-04:
> I thought a bit more about this and in the end haveing the GUI just out of the PCB editor is kind of limited.
Could you explain a little more why you think is not a big deal ?
Thanks for your ideas.

Bouni, 2022-05-04:
What I wanted to say is that a GUI that can just be used out of the PCB editor is not ideal because you may also want to use it from within the schematics editor.
A usual workflow is Schematic -> PCB layout -> Production I guess and jumping back and forth just to use the GUI is not ideal.

With "not a big deal" I meant that it is equaly easy to build a standalone GUI compared to an action plugin.

I did a quick experiment yesterday but that was just a quick and dirty proof of concept so far:

[image]

Here are some of my thoughts / how I did things for my test:

1. Calling `easyeda2kicad` without any parameters starts the GUI, if arguments are provided the tool works in CLI mode
2. These function [__main__.py#L127-L164](https://github.com/uPesy/easyeda2kicad.py/blob/master/easyeda2kicad/__main__.py#L127-L164) should be moved to a seperate file and being imported so that the can be used from the GUI as well
3. The same applies for this logic as it can be used for the GUI as well [__main__.py#L180-L244](https://github.com/uPesy/easyeda2kicad.py/blob/master/easyeda2kicad/__main__.py#L180-L244)

I'll keep you up to date on my experiments!

Bouni, 2022-05-04:
I implemented a very basic version of this proposal, you can have a look at it here: https://github.com/Bouni/easyeda2kicad.py/tree/gui

If this goes into the right direction for you I'll put in some more effort and prepare a PR

uPesy, 2022-05-05:
Thanks for your details explanation, i get it.

> A usual workflow is Schematic -> PCB layout -> Production I guess and jumping back and forth just to use the GUI is not ideal.

Sure, i think the best solution should be to access the plugin in Eeschema and Pcbnew. But it seems that's tricky to create a plugin in Eeschema right ?
For now, having a standalone GUI seems the best approach. It will be also compatible with Kicad v5.x
Your POC GUI is fine for me. The use of `wx` is great for cross platform.
Great work !

uPesy, 2022-05-05:
> If this goes into the right direction for you I'll put in some more effort and prepare a PR

Sure it's seems fine for me.
I have thought a bit more about features that can be added in the gui :

- Having the output log display in the GUI -> users can see if there are warning/errors
- Having all fields values stored in a conf file (or JSON) to prefilled box on next GUI start
- Add a checkbox to select between v5 or v6 lib format
- Add a checkbox to use the default path (user/Documents/Kicad/libs/...) to save converted parts. If it's checked, that disable the custom library path

I'm too busy this week to work on the project, but will definitely work on it next week !

Bouni, 2022-05-05:
At the moment it's not possible to have eeschema plugins! That'll com down the road towards V7 but nobody knows when that's the case ...

Good points you got there!
I'll extend this a bit further until you're happy with it.

Bouni, 2022-06-09:
I keep working on this but progress is slow (wx sucks to be honest 🙈 )

uPesy, 2022-06-09:
Ok
Do you know an alternative GUI lib ? tkinter ?
You use wx in for your plugin right ?

Bouni, 2022-06-09:
Tkinter and Qt come to my mind but I've never used one of them before.

Maybe using a TUI is an option, like https://github.com/bczsalba/pytermgui
But I don't know how that works out on different platforms.

And last but with the most work needed, a web API (maybe fastapi) and a Vue frontend that runs in a browser.

Many options, all with their pros and cons.

Creating a nice looking UI in wx is a pain in the rear but might be the easiest way to do it ...

uPesy, 2022-06-10:
> Creating a nice looking UI in wx is a pain in the rear but might be the easiest way to do it ...

I agree

hattesen, 2023-01-18:
@bouni, would it not be possible for your [kicad-jlcpcb-tools](https://github.com/bouni/kicad-jlcpcb-tools) plugin to show a "Fetch EasyEDA model" button next to a JLPCB component, and when pressed call `easyeda2kicad.py` with appropriate arguments to fetch and convert the EasyEDA model for the component and add to KiCAD libraries?

Bouni, 2023-01-20:
@hattesen This might be possible but see it as a future goal and not a simple low hanging fruit type of thing.

markusdd, 2024-03-10:
@Bouni maybe you are interested in this:
I made a UI now which makes the process more convenient.

Maybe you guys want to check it? I stepped away from doing it in wxwidgets directly in KiCAD because there is not much gain having it in there, but maintaining it is kind of a nightmare.

https://github.com/markusdd/EasyEDA_to_KiCAD_Lib_UI

I released a first windows build, more to follow.

I have not used easyeda2kicad though but JLC2KiCad_lib.

KJ7LNW, 2024-07-12:
See also: https://github.com/yaqwsx/jlcparts/issues/130 for a web-based POC.