uPesy/easyeda2kicad.py #27: Issue importing ID C2891863

ayourk, 2022-05-17:
Great utility.  I've also found [kicad-jlcpcb-tools](https://github.com/Bouni/kicad-jlcpcb-tools) and [JLC2KiCad_lib](https://github.com/TousstNicolas/JLC2KiCad_lib) which are helpful too.  The latter does something similar to this tool, but I don't believe it can get the 3D Models like this one can.  The following shows some tests and errors fetching different parts of this ID:

```
bash> easyeda2kicad --full --overwrite --lcsc_id=C2891863
-- easyeda2kicad.py --
[*] This id is already in /home/user/Documents/Kicad/easyeda2kicad/easyeda2kicad.lib
[*] Creating Kicad symbol library for LCSC id : C2891863
Traceback (most recent call last):
  File "/home/user/.local/bin/easyeda2kicad", line 8, in <module>
    sys.exit(main())
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/__main__.py", line 221, in main
    importer = easyeda_footprint_importer(easyeda_cp_cad_data=cad_data)
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 173, in __init__
    self.output = self.extract_easyeda_data(
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 186, in extract_easyeda_data
    model_3d_name=ee_data_info["3DModel"],
KeyError: '3DModel'

```
```
bash> easyeda2kicad --symbol --overwrite --lcsc_id=C2891863
-- easyeda2kicad.py --
[*] This id is already in /home/user/Documents/Kicad/easyeda2kicad/easyeda2kicad.lib
[*] Creating Kicad symbol library for LCSC id : C2891863

```
```
bash> easyeda2kicad --footprint --overwrite --lcsc_id=C2891863
-- easyeda2kicad.py --
Traceback (most recent call last):
  File "/home/user/.local/bin/easyeda2kicad", line 8, in <module>
    sys.exit(main())
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/__main__.py", line 221, in main
    importer = easyeda_footprint_importer(easyeda_cp_cad_data=cad_data)
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 173, in __init__
    self.output = self.extract_easyeda_data(
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 186, in extract_easyeda_data
    model_3d_name=ee_data_info["3DModel"],
KeyError: '3DModel'
```
```
bash> easyeda2kicad --3d --overwrite --lcsc_id=C2891863
-- easyeda2kicad.py --
[*] Creating 3D model for LCSC id : C2891863
Traceback (most recent call last):
  File "/home/user/.local/bin/easyeda2kicad", line 8, in <module>
    sys.exit(main())
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/__main__.py", line 241, in main
    model_3d=easyeda_3d_model_importer(easyeda_cp_cad_data=cad_data).output
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 254, in __init__
    self.output = self.create_3d_model()
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 262, in create_3d_model
    model_3d: ee_3d_model = self.parse_3d_model_info(
  File "/home/user/.local/lib/python3.8/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 278, in parse_3d_model_info
    name=info["title"],
KeyError: 'title'
```
Here is my version information:
```
Application: KiCad

Version: 6.0.5-a6ca702e91~116~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.13.0-41-generic x86_64, 64 bit, Little endian, wxGTK, , x11

Build Info:
	Date: May  4 2022 07:55:51
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.83.0
	ngspice: 36
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```
I'm running Ubuntu 20.04 at the moment.   I installed this with `pip install` as per the instructions.  Let me know if I should test some sort of dev build.

Hope this helps.  Let me know if you need anything else.

uPesy, 2022-05-17:
Thanks for pointing out this issue.
I think that the id C2891863 does not have 3D model available. That cause the crash. I will add a verification statement.

uPesy, 2022-05-19:
Fix in #29

closed 2022-05-19 by uPesy