uPesy/easyeda2kicad.py #35: Parse error with .kicad_sym in Kicad6

anfractuosity, 2022-05-21:
First of all thanks for this awesome program!

I got a parse issue with kicad6 loading the .kicad_sym file.

I did:
```
easyeda2kicad --full --lcsc_id=C2040 --output ./C2040
```

It generated the following .kicad_sym

```
(kicad_symbol_lib
  (version 20211014)
  (generator https://github.com/uPesy/easyeda2kicad.py
  (symbol "RP2040"
```

There seems to be an issue with no ')' at the end of the generator line.  Adding that seemed to fix the parse error in kicad for me

uPesy, 2022-05-23:
Thanks @anfractuosity for your repot.
Unfornately, i'm not been able to reproduce this errors.
You are on Linux ? which version of easyeda2kicad do you use ? 
Does it appears only one time when a .kicad_sym file is created ? or each times you append a symbol in an existing .kicad_sym ?
It's maybe due to lines end stuf (\r\n) in the .kicad_sym file. These lines append a new symbol in a .kicad_sym .file
https://github.com/uPesy/easyeda2kicad.py/blob/6ac7bae3363c5a5cd185f7e55c24cc9e2b117bbe/easyeda2kicad/helpers.py#L83-L94

anfractuosity, 2022-05-23:
Thanks for your reply. Yep, I'm on Linux, I used the latest version from github.

I tried running 'easyeda2kicad --full --lcsc_id=C2040 --output ./C2040' a few times and got the missing ')' each time.

I'll see if I can look at the code later, to try to work out why I might be getting this

uPesy, 2022-05-23:
Ok
If you change https://github.com/uPesy/easyeda2kicad.py/blob/6ac7bae3363c5a5cd185f7e55c24cc9e2b117bbe/easyeda2kicad/helpers.py#L91 with `lib_file.seek(-1, 2)` does it fix the issue ?

anfractuosity, 2022-05-23:
Yes, that fixes it, thanks :)

uPesy, 2022-05-23:
Ok,
Can you try again with the last commit on the **dev** branch ?
I try to be compatible for both linux and windows
https://github.com/uPesy/easyeda2kicad.py/blob/c38b9392dca68e34659115af3db8067f5676137d/easyeda2kicad/helpers.py#L87-L90

anfractuosity, 2022-05-23:
I pulled the latest changes and checked out 'dev' and uninstalled / re-installed from that git branch.

But unfortunately got  
```
 (generator https://github.com/uPesy/easyeda2kicad.py
```

The commit history I could see had:
```
Date:   Mon May 23 15:22:55 2022 +0200

    try to fix end of lines issue
```

uPesy, 2022-05-24:
Ok, thanks for your feedback
I have made another fix in the dev branch.
Could you try with this new one ? It should be fixed now

uPesy, 2022-05-24:
Fix in PR #38 v0.3.5

closed 2022-05-24 by uPesy