uPesy/easyeda2kicad.py #37: Symbol library failed to load

Bouni, 2022-05-24:
I tried to add `C399482` to my library which seemed to work but it didn't show up in the library.
When I then went to `Preferences -> Manage Symbol librarys` and klicked OK I got his error:

[image]

Heres the line from the error:

```
   (symbol "HFD16/24-ZFH-3N"
```

The `/` seems to be the invalid character so I replaced it with an `_`

Next try another error message:

[image]

```
     (symbol "HFD16/24-ZFH-3N_0_1"
``` 

Again the `/` is the problem, so I replaced it with an `_`

After changing that I no longer got errors from the `Manage Symbol librarys` window but the symbol does not show up in the symbol browser 🤷🏽 

Here's the entire S-Expression of the part:

```
  (symbol "HFD16_24-ZFH-3N"
    (in_bom yes)
    (on_board yes)
    (property
      "Reference"
      "K?"
      (id 0)
      (at 0 15.24 0)
      (effects (font (size 1.27 1.27) ) )
    )
    (property
      "Value"
      "HFD16_24-ZFH-3N"
      (id 1)
      (at 0 -12.70 0)
      (effects (font (size 1.27 1.27) ) )
    )
    (property
      "Footprint"
      "RELAY-TH_HFD16-X-ZXX-XX"
      (id 2)
      (at 0 -15.24 0)
      (effects (font (size 1.27 1.27) ) hide)
    )
    (property
      "Datasheet"
      "https://lcsc.com/product-detail/Relays_Xiamen-Hongfa-Electroacoustic-HFD16-24-ZFH-3N_C399482.html"
      (id 3)
      (at 0 -17.78 0)
      (effects (font (size 1.27 1.27) ) hide)
    )
    (property
      "Manufacturer"
      "Xiamen Hongfa Electroacoustic"
      (id 4)
      (at 0 -20.32 0)
      (effects (font (size 1.27 1.27) ) hide)
    )
    (property
      "LCSC Part"
      "C399482"
      (id 5)
      (at 0 -22.86 0)
      (effects (font (size 1.27 1.27) ) hide)
    )
    (property
      "JLC Part"
      "Extended Part"
      (id 6)
      (at 0 -25.40 0)
      (effects (font (size 1.27 1.27) ) hide)
    )
    (symbol "HFD16_24-ZFH-3N_0_1"
              (rectangle
                (start -10.41 7.62)
                (end 10.16 -5.08)
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type background))
              )
              (polyline
                (pts
                      (xy 2.54 -2.54)
      (xy 2.54 -0.0)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 2.54 2.54)
      (xy 2.54 5.08)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 5.08 2.54)
      (xy 5.08 -0.0)
      (xy 0.0 -0.0)
      (xy 0.0 2.54)
      (xy 5.08 2.54)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type background))
              )
              (polyline
                (pts
                      (xy -7.62 -2.54)
      (xy -7.62 5.08)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 7.62 -2.54)
      (xy 7.62 -0.0)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 7.62 5.08)
      (xy 7.62 2.54)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy -7.62 1.27)
      (xy -3.81 1.27)
      (xy -3.81 -3.81)
      (xy 8.889999999999999 -3.81)
      (xy 8.889999999999999 1.27)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 8.889999999999999 1.27)
      (xy 6.604 -0.762)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 2.54 -2.54)
      (xy 2.54 -0.0)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 2.54 2.54)
      (xy 2.54 5.08)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 5.08 2.54)
      (xy 5.08 -0.0)
      (xy 0.0 -0.0)
      (xy 0.0 2.54)
      (xy 5.08 2.54)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type background))
              )
              (polyline
                (pts
                      (xy -7.62 -2.54)
      (xy -7.62 5.08)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 7.62 -2.54)
      (xy 7.62 -0.0)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 7.62 5.08)
      (xy 7.62 2.54)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy -7.62 1.27)
      (xy -3.81 1.27)
      (xy -3.81 -3.81)
      (xy 8.889999999999999 -3.81)
      (xy 8.889999999999999 1.27)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
              (polyline
                (pts
                      (xy 8.889999999999999 1.27)
      (xy 6.858 3.048)
                )
                (stroke (width 0) (type default) (color 0 0 0 0))
                (fill (type none))
              )
      (pin input line
        (at 2.54 10.16 270)
        (length 2.54)
        (name "2" (effects (font (size 1.27 1.27))))
        (number "2" (effects (font (size 1.27 1.27))))
      )
      (pin input line
        (at 2.54 -7.62 90)
        (length 2.54)
        (name "5" (effects (font (size 1.27 1.27))))
        (number "5" (effects (font (size 1.27 1.27))))
      )
      (pin input line
        (at -7.62 -7.62 90)
        (length 2.54)
        (name "6" (effects (font (size 1.27 1.27))))
        (number "6" (effects (font (size 1.27 1.27))))
      )
      (pin input line
        (at -7.62 10.16 270)
        (length 2.54)
        (name "1" (effects (font (size 1.27 1.27))))
        (number "1" (effects (font (size 1.27 1.27))))
      )
      (pin input line
        (at 7.62 10.16 270)
        (length 2.54)
        (name "3" (effects (font (size 1.27 1.27))))
        (number "3" (effects (font (size 1.27 1.27))))
      )
      (pin input line
        (at 7.62 -7.62 90)
        (length 2.54)
        (name "4" (effects (font (size 1.27 1.27))))
        (number "4" (effects (font (size 1.27 1.27))))
      )
    )
  )
```

Bouni, 2022-05-24:
🤦🏽‍♂️ After I restarted KiCAD the symbol shows up.

But still, you should sanitize the `(symbol "<name>"` part.

uPesy, 2022-05-24:
Yes, I have noticed than unlike in Kicad V5, you have to restart completly Kicad v6 to properly refresh the libraries.
Should be fixed in #38 v0.3.5

Bouni, 2022-05-24:
#38 fixes the / issue but breaks the library in another way.

my symbol gets appended and not inserted:

```
   )
  )
) <---- This is the closing bracket of kicad_symbol_lib
  (symbol "HFD16_24-ZFH-3N"
    (in_bom yes)
    (on_board yes)
```

If I delete that bracket, the library works fine.