uPesy/easyeda2kicad.py #41: Bug: TypeError: expected string or bytes-like object [LCSC ID C264374]

Bouni, 2022-05-30:
I tried to download `C264374` which gave me a `TypeError`:

```
C:\Users\me\Documents\KiCad\6.99.0>easyeda2kicad --full --lcsc_id=C264374
-- easyeda2kicad.py --
[WARNING] Unknow symbol designator : T
[WARNING] Unknow symbol designator : T
[INFO] Creating Kicad symbol library for LCSC id : C264374
[WARNING] No 3D model found for uuid:dce7fcbb60d147f2807f236c5461017e
[INFO] Creating Kicad footprint library for LCSC id : C264374
[INFO] Creating 3D model for LCSC id : C264374
[WARNING] No 3D model found for uuid:dce7fcbb60d147f2807f236c5461017e
Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.99\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\6.99\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\me\Documents\KiCad\6.99.0\3rdparty\Python39\Scripts\easyeda2kicad.exe\__main__.py", line 7, in <module>
  File "C:\Users\me\Documents\KiCad\6.99.0\3rdparty\Python39\site-packages\easyeda2kicad\__main__.py", line 278, in main
    exporter = Exporter3dModelKicad(
  File "C:\Users\me\Documents\KiCad\6.99.0\3rdparty\Python39\site-packages\easyeda2kicad\kicad\export_kicad_3d_model.py", line 119, in __init__
    self.output = generate_wrl_model(model_3d=model_3d) if model_3d else None
  File "C:\Users\me\Documents\KiCad\6.99.0\3rdparty\Python39\site-packages\easyeda2kicad\kicad\export_kicad_3d_model.py", line 54, in generate_wrl_model
    materials = get_materials(obj_data=model_3d.raw_obj)
  File "C:\Users\me\Documents\KiCad\6.99.0\3rdparty\Python39\site-packages\easyeda2kicad\kicad\export_kicad_3d_model.py", line 22, in get_materials
    matchs = re.findall(pattern=material_regex, string=obj_data, flags=re.DOTALL)
  File "C:\Program Files\KiCad\6.99\bin\lib\re.py", line 241, in findall
    return _compile(pattern, flags).findall(string)
TypeError: expected string or bytes-like object
```

uPesy, 2022-05-31:
Thanks for report this.
You run easyeda2kicad through Kicad ? Do you have also installed using pip with the Kicad python binary ?

Bouni, 2022-05-31:
Yes indeed, I use the Kicad CMD prompt which uses the python shipped with Kicad

uPesy, 2022-05-31:
Should be fixed in PR #42

Bouni, 2022-06-01:
Now it works 👍🏽

closed 2022-06-01 by Bouni