uPesy/easyeda2kicad.py #45: Add support for circles and arcs in symbols

Bouni, 2022-06-09:
By now circles and arcs are not supported which leads to wierd symbles sometimes, here's nan example for `C318884`

Easyeda symbol:
[image]

KiCAD symbol generated by this tool:
[image]

According to the code, circles and arcs are not yet implemented:

- https://github.com/uPesy/easyeda2kicad.py/blob/6ac7bae3363c5a5cd185f7e55c24cc9e2b117bbe/easyeda2kicad/easyeda/parameters_easyeda.py#L143-L152

Bouni, 2022-06-13:
Nice, circles seem to work in the latest dev commit 😍 

[image]

[image]

uPesy, 2022-06-13:
Yes but i'm not sure if i should enable filling the background (having the circle with yellow background)
The circle are descrided as ellipse in the easyeda footprint 🤔

Bouni, 2022-06-14:
Does the easyeda data show if their symbol has filled circles or not?

uPesy, 2022-06-14:
No i don't think so

Bouni, 2022-06-14:
In that case I would no fill them in ...

uPesy, 2022-06-15:
Yeah you're right, but in easyeda, there is no background filling on symbol, contrary to Kicad :
[image]
[image]

Bouni, 2022-06-15:
For `C318884`: background fill in circles is gone, looks good to me!

[image]

The optocoupler issue is answered in #53

uPesy, 2022-06-19:
Symbol arc is now supported in the last version v0.5.0.
Note that easyeda use elleptical arc while Kicad support only circle arc.

Bouni, 2022-06-20:
Are you aware of a symbol that use actual elliptical arcs?

I mean they most likely use elliptical arcs as "normal" arcs but never as an elliptical shape, right?

uPesy, 2022-06-20:
For the coil, the radius is `rx=3.9` and `ry=4.0` on Easyeda

> I mean they most likely use elliptical arcs as "normal" arcs but never as an elliptical shape, right?
Yes

Bouni, 2022-06-20:
What do you plan to do in such cases? 
I could imagine having a threshold for the difference in `rx` and `ry` of lets say 0.1mm and if that threshold is not exceeded use one of the values as arc radius. That way we end up wit usable symbols that might not fit a 100% but still better than skip the arcs completely.

Bouni, 2022-06-20:
Maybe easier as my suggestion. Thinks that'll do the job.