uPesy/easyeda2kicad.py #46: Found another edge case!  LCSC part C470643

ayourk, 2022-06-09:
Maybe missing info for this part?  There is an EasyEDA link in the listing.
Here is the listing: https://lcsc.com/product-detail/Variable-Resistors-Potentiometers_ALPSALPINE-RK10J11E0034_C470643.html
```
ayourk@Z590-P:~/Documents/Kicad>~/.local/bin/easyeda2kicad --symbol --lcsc_id=C470643
-- easyeda2kicad.py v0.3.7 --
Traceback (most recent call last):
  File "/home/ayourk/.local/bin/easyeda2kicad", line 8, in <module>
    sys.exit(main())
  File "/home/ayourk/.local/lib/python3.10/site-packages/easyeda2kicad/__main__.py", line 221, in main
    importer = EasyedaSymbolImporter(easyeda_cp_cad_data=cad_data)
  File "/home/ayourk/.local/lib/python3.10/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 14, in __init__
    ee_data_info=easyeda_cp_cad_data["dataStr"]["head"]["c_para"],
KeyError: 'dataStr'
ayourk@Z590-P:~/Documents/Kicad>~/.local/bin/easyeda2kicad --footprint --lcsc_id=C470643
-- easyeda2kicad.py v0.3.7 --
Traceback (most recent call last):
  File "/home/ayourk/.local/bin/easyeda2kicad", line 8, in <module>
    sys.exit(main())
  File "/home/ayourk/.local/lib/python3.10/site-packages/easyeda2kicad/__main__.py", line 259, in main
    importer = EasyedaFootprintImporter(easyeda_cp_cad_data=cad_data)
  File "/home/ayourk/.local/lib/python3.10/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 159, in __init__
    ee_data_str=self.input["packageDetail"]["dataStr"],
KeyError: 'packageDetail'
ayourk@Z590-P:~/Documents/Kicad>~/.local/bin/easyeda2kicad --3d --lcsc_id=C470643
-- easyeda2kicad.py v0.3.7 --
[INFO] Creating 3D model for LCSC id : C470643
Traceback (most recent call last):
  File "/home/ayourk/.local/bin/easyeda2kicad", line 8, in <module>
    sys.exit(main())
  File "/home/ayourk/.local/lib/python3.10/site-packages/easyeda2kicad/__main__.py", line 279, in main
    model_3d=Easyeda3dModelImporter(
  File "/home/ayourk/.local/lib/python3.10/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 240, in __init__
    self.output = self.create_3d_model()
  File "/home/ayourk/.local/lib/python3.10/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 244, in create_3d_model
    self.input["packageDetail"]["dataStr"]["shape"]
KeyError: 'packageDetail'
ayourk@Z590-P:~/Documents/Kicad>~/.local/bin/easyeda2kicad --full --lcsc_id=C470643            
-- easyeda2kicad.py v0.3.7 --
Traceback (most recent call last):
  File "/home/ayourk/.local/bin/easyeda2kicad", line 8, in <module>
    sys.exit(main())
  File "/home/ayourk/.local/lib/python3.10/site-packages/easyeda2kicad/__main__.py", line 221, in main
    importer = EasyedaSymbolImporter(easyeda_cp_cad_data=cad_data)
  File "/home/ayourk/.local/lib/python3.10/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 14, in __init__
    ee_data_info=easyeda_cp_cad_data["dataStr"]["head"]["c_para"],
KeyError: 'dataStr'
ayourk@Z590-P:~/Documents/Kicad>
```

Bouni, 2022-06-10:
I just looked at this out of interest, for this part there is no symbol nor footprint available in easyeda:

[image]

With the logging of easyeda2kicad set to debug it is clear that the API returns a 404 for this part number but that is not handled correctly:

```
-- easyeda2kicad.py v0.3.7 --
[INFO] Create easyeda2kicad.kicad_sym symbol lib in C:\Users\me\Documents\Kicad\easyeda2kicad
[DEBUG] Starting new HTTPS connection (1): easyeda.com:443
[DEBUG] https://easyeda.com:443 "GET /api/products/C470643/components?version=6.4.19.5 HTTP/1.1" 200 None

[DEBUG] {'success': False, 'code': 404, 'message': 'Component not found'}

Traceback (most recent call last):
  File "C:\Users\me\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\me\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Projekte\easyeda2kicad.py\easyeda2kicad\__main__.py", line 288, in <module>
    sys.exit(main(sys.argv[1:]))
  File "c:\Projekte\easyeda2kicad.py\easyeda2kicad\__main__.py", line 221, in main
    importer = EasyedaSymbolImporter(easyeda_cp_cad_data=cad_data)
  File "c:\Projekte\easyeda2kicad.py\easyeda2kicad\easyeda\easyeda_importer.py", line 14, in __init__
    ee_data_info=easyeda_cp_cad_data["dataStr"]["head"]["c_para"],
KeyError: 'dataStr'
```

uPesy, 2022-06-10:
Should be fixed in version 0.4.0 with PR #48

PR #47 (2022-06-10): Handle LCSC API 404 correct
This fixes #46
closed 2022-06-10 by uPesy