uPesy/easyeda2kicad.py #53: Missing lines in C6649 / C2128 symbol

Bouni, 2022-06-14:
I just discoverd that there's one line missing for the diode in the optocoupler `C6649` as well as the diode `C2128`:

[image]

[image]

[image]

[image]

uPesy, 2022-06-15:
Should be fixed in version 0.4.2 on the dev branch

Bouni, 2022-06-15:
`C6649` works for me in 0.4.2 but `C2128` is still the same:

[image]

[image]

uPesy, 2022-06-18:
On my side it's working great too for `C2128` 
[image]

uPesy, 2022-06-19:
In fact, it's because the lib update features doesn't work with this particular ID with `(` and `)` in the regex.
It's fixed in the last version on the dev branch.

Bouni, 2022-06-20:
:tada: Works for me now!

closed 2022-06-20 by Bouni