uPesy/easyeda2kicad.py #55: Validation Error for C90272

Bouni, 2022-06-16:
For [C90272](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C90272) I get a validation error with the latest dev version:

```
-- easyeda2kicad.py v0.4.2 --
Traceback (most recent call last):
  File "C:\Users\user7\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\user7\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Projekte\easyeda2kicad.py\easyeda2kicad\__main__.py", line 326, in <module>
    sys.exit(main(sys.argv[1:]))
  File "c:\Projekte\easyeda2kicad.py\easyeda2kicad\__main__.py", line 258, in main
    importer = EasyedaSymbolImporter(easyeda_cp_cad_data=cad_data)
  File "c:\Projekte\easyeda2kicad.py\easyeda2kicad\easyeda\easyeda_importer.py", line 111, in __init__
    self.output: EeSymbol = self.extract_easyeda_data(
  File "c:\Projekte\easyeda2kicad.py\easyeda2kicad\easyeda\easyeda_importer.py", line 139, in extract_easyeda_data
    easyeda_handlers[designator](line, new_ee_symbol)
  File "c:\Projekte\easyeda2kicad.py\easyeda2kicad\easyeda\easyeda_importer.py", line 91, in add_easyeda_arc
    EeSymbolArc(**dict(zip(EeSymbolArc.__fields__, arc_data.split("~")[1:])))
  File "pydantic\main.py", line 406, in pydantic.main.BaseModel.__init__
pydantic.error_wrappers.ValidationError: 1 validation error for EeSymbolArc
helper_dots
  value is not a valid list (type=type_error.list)
```

This is the shape data from the API:

```json
"shape": [
        "P~show~0~1~0~-20~270~gge2~0^^0~-20^^M 0 -20 v 17~#8D2323^^0~3~0~270~1~end~~~#8D2323^^0~-1~-9~270~1~start~~~#8D2323^^0~0~0^^0~M 3 -3 L 0 -6 L -3 -3",
        "P~show~0~2~0~20~90~gge9~0^^0~20^^M 0 20 v -19~#8D2323^^0~3~-2~270~2~start~~~#8D2323^^0~-1~7~270~2~end~~~#8D2323^^0~0~-2^^0~M -3 1 L 0 4 L 3 1",
        "PL~-10 -3 10 -3~#8D2323~1~0~none~gge18~0",
        "A~M 0 1 A 16.0836 16.0836 0 0 1 10 4~~#8D2323~1~0~none~gge19~0",
        "A~M -10 4 A 16.2917 16.2917 0 0 1 0 1~~#8D2323~1~0~none~gge20~0",
        "R~-10~-6~~~4~0.1~#8D2323~1~0~#8D2323~gge21~0~",
        "R~-8~-8~~~0.1~4~#8D2323~1~0~#8D2323~gge22~0~"
      ]
```

I guess that this ARC is the problem:

[image]

Bouni, 2022-06-16:
The same error is thrown for [C149588](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C149588)

I think here are the 4 ARCs responsible for the error:

[image]

uPesy, 2022-06-19:
Should be fixed in the last version in the dev branch v0.5.0

Bouni, 2022-06-20:
`C149588` works fine now:

[image]

But `C90272` has missing lines and flipped pins:

[image]

uPesy, 2022-06-20:
Could you open another issue and close this one ?
It's due to the pin length. Until now, i have fixed the same pin length for all symbols.

closed 2022-06-22 by uPesy