uPesy/easyeda2kicad.py #69: C398355 - fails to import symbol

bzed, 2022-09-20:
C398355 is a dual opamp, its symbol on easyeada comes in two(?) parts, the import basically imports an empty symbol.

There is a version of the chip in kicad, there you have to add three symbols and end up with more than one footpriunt that needs to be aligned. Both versions are not nice :)

uPesy, 2022-09-21:
Yeah, i don't think that the current version of easyeda2kicad support to convert a symbol that is split in severals parts. I have not manage this case :/

bzed, 2022-11-21:
Stale, but I think worth fixing it.

uPesy, 2022-11-21:
I don't have the time to implement this right now sorry

fpovoski, 2022-12-01:
Suddenly(as in this morning), after importing numerous symbols into easyeda2kicad.kicad_sym, any further imports the program completes without errors, yet no symbol is actually populated in the kicad_sym file. Any ideas why?