uPesy/easyeda2kicad.py #71: Fails to import

fpovoski, 2022-12-01:
Suddenly(as in this morning), after importing numerous symbols into easyeda2kicad.kicad_sym, any further imports the program completes without errors, yet no symbol is actually populated in the kicad_sym file. Any ideas why?

fpovoski, 2022-12-01:
In case you want to see debug:
C:\Users\fpovo>easyeda2kicad --symbol --overwrite --debug --lcsc_id C6186
-- easyeda2kicad.py v0.6.2 --
[DEBUG] Starting new HTTPS connection (1): easyeda.com:443
[DEBUG] https://easyeda.com:443 "GET /api/products/C6186/components?version=6.4.19.5 HTTP/1.1" 200 None
[INFO] Created Kicad symbol for ID : C6186
       Symbol name : AMS1117-3.3
       Library path : C:\Users\fpovo\Documents\Kicad\easyeda2kicad/easyeda2kicad.kicad_sym

fpovoski, 2023-12-17:
Same thing a year later.  Imports footprint and 3d model but will not fetch the symbol.  Have to create libraries other than "easyeda2kicad/easyeda2kicad.kicad_sym", which is annoying. Can you fix this?

binary1230, 2024-01-04:
maybe it's this? https://github.com/uPesy/easyeda2kicad.py/issues/87

uPesy, 2026-04-06:
Fix in #184