uPesy/easyeda2kicad.py #72: installing through windows doesnt work

Cowh3adDK, 2022-12-14:
after installing the package, running easyeda2kicad returns "easyeda2kicad : The term 'easyeda2kicad' is not recognized as the name of a cmdlet, function, script file, or operable
program."

Cowh3adDK, 2022-12-14:
i think somewhere something gets confused if you have wsl ubuntu installed with python, uninstalled it and reinstalled python and easyeda2kicad on windows and it works now.

Jackowski3k, 2023-01-13:
Will it work if you call it by "python -m easyeda2kicad"? Probably windows installation not providing the modules dir to system PATH?

Cowh3adDK, 2023-01-13:
The problem was python wasnt added to my path, I had it installed from
docker I think, just uninstalled docker and python and installed python
manually.

uPesy, 2023-01-19:
A clever way to install it without installing python is using the Kicad Command Prompt that have Python already install on it

closed 2023-01-19 by uPesy