uPesy/easyeda2kicad.py #73: Minor issue with through hole pads

Bouni, 2022-12-19:
I've downloaded [C77837](https://jlcpcb.com/partdetail/ConnflyElec-DS1037_15MNCKT740CC/C77837), a D-SUB connector.
The holes look like they are filled with copper:

[image]

But wehn I open them in the editor, change the type from THT to SMD and back, the appear right.

On closer inspection I've seen that they all have `*.Paste` in their layers, bcause the part is set to SMD:

[image]

But changing that to THT doesn't solve the issue, I've to remove the `*.Paste` from every pad in a texte ditor:

```
  (pad "" thru_hole circle (at -16.53 0) (size 4 4) (drill 3.3) (layers *.Cu *.Mask) (tstamp a98297e3-cdc7-4872-ad2f-4ada1c6ce0e9))
  (pad "" thru_hole circle (at 16.79 0) (size 4 4) (drill 3.3) (layers *.Cu *.Mask) (tstamp 2b1a6742-90de-45cd-acf2-2c3211d682ea))
  (pad "1" thru_hole circle (at 9.73 -1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp 6a7ed235-d9c5-4d2b-bade-77126e580c77))
  (pad "2" thru_hole circle (at 6.95 -1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp df04db4d-e4a9-4cf8-844d-f61d545a4416))
  (pad "3" thru_hole circle (at 4.17 -1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp ddf7d6f8-48ad-42cd-b128-d7fb18c1a735))
  (pad "4" thru_hole circle (at 1.39 -1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp 3af50f33-530e-4e89-935c-a083c9a4c41f))
  (pad "5" thru_hole circle (at -1.39 -1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp 4cc5cb1d-adc3-464c-8d52-fbe477e18693))
  (pad "6" thru_hole circle (at -4.17 -1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp bdc8f212-b8a7-4082-ad82-0a85f5085f2d))
  (pad "7" thru_hole circle (at -6.95 -1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp 2a332a1d-289b-4303-8172-b62883e267d8))
  (pad "8" thru_hole circle (at -9.73 -1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp 1071132d-245e-4980-950c-3c2fa136c68b))
  (pad "9" thru_hole circle (at 8.34 1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp fdd17afd-52cc-463b-bd25-57d4e67b9840))
  (pad "10" thru_hole circle (at 5.56 1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp 95192e95-2a70-49e9-a255-5ac9b614511b))
  (pad "11" thru_hole circle (at 2.78 1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp f3f6d7a7-6dd8-456b-8607-c05a3ca4af77))
  (pad "12" thru_hole circle (at 0 1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp 0227ef85-4dd4-41ca-a192-87bdc7fb9ca8))
  (pad "13" thru_hole circle (at -2.78 1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp 09cc6263-b3ee-43c3-9d0b-b9d54dc804ab))
  (pad "14" thru_hole circle (at -5.56 1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp c7942b37-e066-4353-a2bd-5cf3df2bb8dc))
  (pad "15" thru_hole circle (at -8.34 1.42 180) (size 1.8 1.8) (drill 1.199896) (layers *.Cu *.Mask) (tstamp f0220c46-fb05-47b0-8264-869ddd6623e9))
```

This solves the issue but is very labor intensive ...

[image]

fdebrabander, 2023-01-18:
Having the same issue with all components that have through hole pins. In the 3d model you can clearly see that the components don't look right. Can be manually fixed with

1. Open item in footprint editor
2. Double-click on through-hole object (yellow circle)
3. Remove check mark from F.Paste and B.Paste

netweaver1970, 2023-03-31:
Root cause it at the Easyeda lib side ... See my comment: https://github.com/uPesy/easyeda2kicad.py/issues/91#issuecomment-1491911561

netweaver1970, 2023-04-01:
fixed here: https://github.com/uPesy/easyeda2kicad.py/issues/91#issuecomment-1492881348

uPesy, 2026-04-06:
Fix in #184

closed 2026-04-06 by uPesy