uPesy/easyeda2kicad.py #75: Conflicts with parts in other libraries

Jackowski3k, 2023-01-16:
I'm not sure if I'm doing something wrong, but I noticed, that when I want to download a part that already exist in other libraries, it doesn't want to download the part and asks to overwrite - then with --overwrite flag it still doesn't download it to the default library.

The solution I found is to explicitly provide the output parameter - then all parts land in the separate library I created for the easyeda2kicad parts.

Jackowski3k, 2023-01-30:
Sorry for long delay. 

I tried reproducing the prolem on other machine - without luck. 
The problem that I experienced earlier - using the default lib (no output parameter in cmdline), I got messages "already exist, use --overwrite" for parts that didn't existed in that lib (but did existed in other lib that was used in project.

There is high chance I did something wrong so you can close the issue here. I'll post update once/if I can reproduce the problem reliably.

uPesy, 2026-04-06:
Fix in #184