uPesy/easyeda2kicad.py #80: 3D models: Problems with the `d` interpretation (dissolve vs transparency)

set-soft, 2023-01-29:
The current code interprets the `d` attribute in raw_obj as transparency, but looking at some [references](https://www.fileformat.info/format/material/) this seems to be another thing called dissolve, which is related to transparency, but isn't the same.

The part **C115366** (SW-TH_SPEF110100) has `d=1` and generates a WRL with transparency 1, making the whole object invisible.

I think the exporter should avoid using the d parameter. Do you have examples where d is used in the same way as transparency? I tried with various LEDs with transparent lenses and all used d=0.

uPesy, 2023-01-29:
Hi @set-soft ,
Thanks for your report.
You can create a PR that implements this feature

set-soft, 2023-01-30:
Ok, I created #81