uPesy/easyeda2kicad.py #82: [Feature request] How about exporting as STEP

set-soft, 2023-01-30:
Hi!
The scripts are currently generating a WRL, but KiCad 3D models must be in STEP and WRL.
The WRL is used to render the board, but the STEP is needed to export the board as a 3D model to validate distances.
I know STEP files are solids and WRL files are just meshes, but may be you know about some mechanism to get something usable.

P.S. The original format seems to be OBJ, which looks like a simple mesh ...

uPesy, 2023-01-30:
Yes i know, but it's difficult to convert a WRL to STEP
Any idea to this from the OBJ or WRL will be appreciated :)

darkxst, 2023-03-10:
Easyeda Pro can export step models of LCSC components, atleast manually. Not sure if that is something thats available through the API though.

uPesy, 2023-03-10:
Yes, Easyeda Pro can do that. But it's a different API with user authentification.
Moreover, it's a paid feature, so i will not offer a way for having this for free.
An independant converter module seems a better solution.

set-soft, 2023-03-10:
I think they have the STEP files and what we get from the free API are files generated from the STEP files.

netweaver1970, 2023-03-30:
> 

What if we have a licence to Easyeda Pro (still free these days), wouldn't it be possible to call that 'different' API using our entered credentials? Maybe at runtime decide to call the free API if no user Auth is provided or use the more capable API if all is provided? Then we could mimic being the app. Or is that API app key (and not user specific auth) based?

Ing-Dom, 2023-10-18:
f"https://modules.easyeda.com/qAxj6KHrDKw4blvCG8QJPs7Y/{component_uuid}"

PR #118 (2024-02-19): Add STEP 3D importer
This adds support for importing 3D models in the STEP format.

Closes #82.
closed 2024-02-19 by uPesy