uPesy/easyeda2kicad.py #87: Adding Symbol fails

danielh186, 2023-03-23:
After editing a symbol with the Kicad Symbol Editor the script can't add symbols to the library anymore.
There is no error shown by the script but KiCAD can't find the symbol inside the library. When I try to add the symbol a second time, an error is thrown ("This id is already in ...").

My workaround for now is just running the script on a symbol library that I don't edit and then importing the symbol into my real library using the Symbol Editor. But I would prefer not having to do this extra step.

The error can be replicated as follows:
1. Create a clean symbol library
2. Add components to the library with the script (works just fine)
3. Edit any component in the symbol library using the KiCAD Symbol Editor (for example change Pin Type from "Unspecified" to "Input"
4. Try to add another component using the script

sogaard, 2023-08-12:
I noticed the same problem.
I looked closer at the library file, and without fully knowing the sym syntax, it looks like a parenthesis mismatch issue. The new symbol was inserted outside the closing parenthesis.
When I manually deleted the additional closing parenthesis in the middle, it sees the new parts.

dgbalharrie, 2023-10-25:
I must have edited a symbol with the kicad editor because new LCSC components were not appearing in the symbol library. Afer reading this issue, I opened the easyedatokicad sym file in a text editor and found an extra bracket half way down the file.
[image]

Once I removed this bracket ( the one in red) all the components in the file beyond the removed bracket appeared in the library list.

binary1230, 2024-01-04:
I ran into the same issue just now too with importing C53406

I had a library with 3 symbols I imported successfully with easyeda2kicad

When I imported the 4th, it inserted an extra paren in the middle of the file (like what was shown before).

The command I was using was:
easyeda2kicad --lcsc_id=C53406 --project-relative --full --output .\LCSC-imported.kicad_sym --overwrite

When I manually removed the extra paren (as highlighted in red by a code editor), that fixed it right up.

---

I'll try and build a more concise repro if that's useful. I hit this a lot and this is the first time I happened to track down what happened.

I wonder if an easy thing to do to track this down is just to have easyeda2kicad validate the file format after the import but before it writes it to the disk, and throw an error if the validation fails.

Then you could get a bug report with the before and after.

If I get a minute I'll try and look into it a little if there's not a known fix.

This was happening on v0.6.3, and still happening when I upgraded to latest (0.6.6)

---

(btw THANK YOU for this amazing tool, it's such a timesaver)

binary1230, 2024-01-04:
Oh yea and,

When this happened, I had 4 symbols in my .kicad_sym file.  However, the position of the bad paren was in the middle of the file after the 1st symbol.

Kicad appeared to be happy reading that first symbol but ignored all the other symbols after that point.  (I guess when the final paren is closed in the file, it just ignores  all the other stuff after that point. I mean.. what else can it do I guess?)

binary1230, 2024-01-04:
I THINK I FOUND IT and have good repro.  It's related to the file ending in a newline or not and LF vs CRLF.

Repro:
1. Make a file named '.\LCSC-imported.kicad_sym' and ensure:
- it is CRLF (windows format) 
- with the following contents (Important: BE SURE IT includes the BLANK NEWLINE AT THE END)
- (this is just a minimal example enough to repro the bug)

```
(kicad_symbol_lib (version 20220914) (generator kicad_symbol_editor)
  (symbol "TXB0108PWR" (in_bom yes) (on_board yes)
    (symbol "TXB0108PWR_1_1"
      (text "Low\nVoltage\nSide" (at -2.54 12.7 0)
        (effects (font (size 0.4 0.4)))
      )
    )
  )
)

```

line 10 should be a newline
[image]

3. run the following command:
easyeda2kicad --lcsc_id=C2987682 --project-relative --output .\LCSC-imported.kicad_sym --symbol

EXPECTED:
new symbol is imported and library works

ACTUAL:
new symbol is imported after the closing parenthesis, and the import silently fails. KiCad will never recognize any imports from this file ever again.

----

Probably the key to the whole thing is, if you remove the newline so the last character is a parenthesis, everything works.

To repro that:

1. take the same file but this time remove the newline at the end:
```
(kicad_symbol_lib (version 20220914) (generator kicad_symbol_editor)
  (symbol "TXB0108PWR" (in_bom yes) (on_board yes)
    (symbol "TXB0108PWR_1_1"
      (text "Low\nVoltage\nSide" (at -2.54 12.7 0)
        (effects (font (size 0.4 0.4)))
      )
    )
  )
)
```

note that there's no newline at the end this time:
[image]

3. run the same easyeda2kicad command above

EXPECTED and ACTUAL: imports successfully.

---

finally, the repro above does NOT trigger the bug IF the file is Unix linendings (LF)

I see the code is doing a bunch of matching of "\n" directly, maybe have to convert to python's lineseparator

I hope that helps. I'll try and look into it more if I have time tomorrow. Thanks!

channelsixlabs, 2024-06-17:
> I THINK I FOUND IT and have good repro. It's related to the file ending in a newline or not and LF vs CRLF.

Good work. Tracking this down in my symbols library has convinced the import script to begin working again.

Hopefully this can be addressed so it can't happen again. Perhaps some post-import step in the script?

binary1230, 2024-06-17:
I think there's probaby a silly but probably effective way to fix this which is, The first thing the script should do when reading the existing symbol file is convert all windows line endings to Unix. It seems like this would be a good thing to do regardless of whatever platform the script is actually running on to ensure line ending consistency

Then I bet everything else would work just fine. It is likely because the python script is doing matching on \n and not also \r\n that we are running into this. 

The only caveat would be if this would break anything because it changes fields inside quoted strings and stuff like that. So at the end of the day this might have to be done at a more structured level that is aware of how things are parsed in KiCads symbol library format.

I suspect it would be good enough though and maybe including a command line to disable the automatic reformatting from Windows to Unix file and things would probably be sufficient if anybody runs into weird edge cases.

uPesy, 2024-06-17:
@binary1230 A PR to fix this issue will be very apreciated.

binary1230, 2024-06-17:
No prob, I'll probably be doing some more electronics design work in the
near future and will try and knock it out as part of that

Blaster1920, 2025-02-05:
As of today, I can confirm this is still an issue.
Thanks @binary1230 for the fix.

joesphan, 2025-04-20:
I am seeing the same issue. Have to open in text editor every time it doesn't work

dgbalharrie, 2025-06-14:
There is another way to import symbols and footprints. In LCSC, bring up the component details, then look for EasyEda Models. 

[image]

Click the Free Trial model. This will open EasyPro with two tabs, one is the symbol and the other tab is the footprint. From each tab use File/Save As/Local File

[image]

This will download two files to your Downloads folder.

From Kicad use import in the Symbol Editor and then Footprint editor. In each editor select the library you want to import to, then from import select the appropraite file from your download folder.

You will need to use this software to import the 3D symbol.

dgbalharrie, 2025-06-16:
> Yes you can also batch export EasyEDA components to ezlib format, which can freely be imported into KiCad, but if this tool worked, it would be far easier...

That might be true, but someone has to find the time to address this issue… It has been around for a long time.

Mcneelsh, 2025-08-07:
> That might be true, but someone has to find the time to address this issue… It has been around for a long time.
> 
> From: Shaun Wild ***@***.***>
> Sent: 16 June 2025 15:22
> To: uPesy/easyeda2kicad.py ***@***.***>
> Cc: David Balharrie ***@***.***>; Comment ***@***.***>
> 
> [Image removed by sender.]shaun-wild left a comment ([uPesy/easyeda2kicad.py#87](https://github.com/uPesy/easyeda2kicad.py/issues/87))<[#87 (comment)](https://github.com/uPesy/easyeda2kicad.py/issues/87#issuecomment-2976857029)>
> 
> Yes you can also batch export EasyEDA components to ezlib format, which can freely be imported into KiCad, but if this tool worked, it would be far easier...

I'm not by any stretch a programmer, but I'm going to take a look this weekend and see if I cant get this issue fixed, even if its just a patch job. Im working on developing a course for my uni where we would like to offer students the ability to use this tool should they need it.