uPesy/easyeda2kicad.py #88: Adding a part with a colon in the name causes creation of invalid symbol

itsumimario, 2023-03-24:
When adding a part that has a colon in the name, the script creates a symbol that Kicad doesn't like, and will give an error message saying something like "invalid prefix at line 56 offset 14". For example, part C26393 will do this. 

The solution is to substitute "{colon}" for ":" in the symbol name.

For example, here is the invalid symbol that the script generates:
(kicad_symbol_lib (version 20220914) (generator kicad_symbol_editor)
  (symbol "MT48LC16M16A2P-6AIT:G" (in_bom yes) (on_board yes)
    (property "Reference" "U" (at 0 38.1 0)
      (effects (font (size 1.27 1.27)))
    )
    (property "Value" "MT48LC16M16A2P-6A IT:G" (at 0 -38.1 0)
      (effects (font (size 1.27 1.27)))
    )
    (property "Footprint" "easyeda2kicad:TSOP-54_L22.2-W10.2-P0.80-LS11.8-BL" (at 0 -40.64 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (property "Datasheet" "https://lcsc.com/product-detail/RAM_Micron_MT48LC16M16A2P-6AIT-G_MT48LC16M16A2P-6AIT-G_C26393.html" (at 0 -43.18 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (property "Manufacturer" "micron(镁光)" (at 0 -45.72 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (property "LCSC Part" "C26393" (at 0 -48.26 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (property "JLC Part" "Extended Part" (at 0 -50.8 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (symbol "MT48LC16M16A2P-6AIT:G_0_1"
      (rectangle (start -8.89 35.56) (end 8.89 -35.56)
        (stroke (width 0) (type default))
        (fill (type background))
      )
      (circle (center -7.62 34.29) (radius 0.38)
        (stroke (width 0) (type default))
        (fill (type none))
      )
... blah blah blah
    )
  )
)

But it will work if you substitute two of the ":" instances with "{colon}" like this:

(kicad_symbol_lib (version 20220914) (generator kicad_symbol_editor)
  (symbol "MT48LC16M16A2P-6AIT{colon}G" (in_bom yes) (on_board yes)
    (property "Reference" "U" (at 0 38.1 0)
      (effects (font (size 1.27 1.27)))
    )
    (property "Value" "MT48LC16M16A2P-6A IT:G" (at 0 -38.1 0)
      (effects (font (size 1.27 1.27)))
    )
    (property "Footprint" "easyeda2kicad:TSOP-54_L22.2-W10.2-P0.80-LS11.8-BL" (at 0 -40.64 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (property "Datasheet" "https://lcsc.com/product-detail/RAM_Micron_MT48LC16M16A2P-6AIT-G_MT48LC16M16A2P-6AIT-G_C26393.html" (at 0 -43.18 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (property "Manufacturer" "micron(镁光)" (at 0 -45.72 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (property "LCSC Part" "C26393" (at 0 -48.26 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (property "JLC Part" "Extended Part" (at 0 -50.8 0)
      (effects (font (size 1.27 1.27)) hide)
    )
    (symbol "MT48LC16M16A2P-6AIT{colon}G_0_1"
      (rectangle (start -8.89 35.56) (end 8.89 -35.56)
        (stroke (width 0) (type default))
        (fill (type background))
      )
      (circle (center -7.62 34.29) (radius 0.38)
        (stroke (width 0) (type default))
        (fill (type none))
      )
... blah blah blah
    )
  )
)

boromyr, 2023-09-24:
I also encounter this problem

uPesy, 2026-04-06:
Fix in #184

PR #146 (2025-02-11): Escape ":" in symbol names
Fixes #88

I was importing C2859183, which has a symbol named "MT40A1G8SA-062E:R" when I ran into this problem, so I wrote this simple fix.

I also fixed an oversight in the duplicate detection that I found when testing, where if ":" was in the symbol name it would not find the existing symbol. This presumably could have happened with " " and "/" too as they are also replaced in symbol names.
closed 2026-04-06 by uPesy