uPesy/easyeda2kicad.py #91: 'floating 3d parts'in Kicad after full import from Easyeda

netweaver1970, 2023-03-30:
Hi,
I tried to manually recreate a small board in Kicad 7.0.1, after having created it in Easyeda Pro.
To ease the task, I used e2k to import ALL my used parts from Easyeda Pro, even the transistors/passives.
The symbols and footprints all worked fine, but the 3d objects were a bit hit-and-miss.

I'm already grateful I don't have to recreate all myself but seeing the issues, there might be a systemic issues in certain cases (PTH?), causing this import behaviour, so I decided to make an issue, to let you know what might need a look.

This is the result in Easyeda Pro:
[image]

This is the result in Kicad:
[image]

The 'Floating up' (only Z height wrong in 3d model) parts are:

- C77859 (the PTH RJ12 Female connector)
- C2932672 (the PTH 7pin Female pin header)

The 'drifting' part (Z height correct but X/Y offset incorrect in 3d model)

- C503582 (PTH ESP-01S)

Missing 3d model all together in Kicad for this part:

- C2145 (MMBT5551 NPN BJT, SMD SOT-23-3)

All parts (with perfect and not-so-good/missing 3d models) were imported one after another using the same script.

Thanks for having a look and all your work on this script !

netweaver1970, 2023-03-31:
It seems this change [https://github.com/uPesy/easyeda2kicad.py/commit/556aa626e1ac3ec712ba6b664ce33b07d1748555](https://github.com/uPesy/easyeda2kicad.py/commit/556aa626e1ac3ec712ba6b664ce33b07d1748555) fix to the code is not doing what it was supposed to do: the is_smd attribute is not filled with 'false' for tht components. 
I have no clue yet why, my python skills are limited so far to ISS level reading ... 
Maybe this is a sign to get myself a bit deeper into it....

netweaver1970, 2023-03-31:
Digging a bit deeper, the root cause is on the Easyeda side, below is the result from the footprint API call. One can see that the SMT flag is set to true. And that's why the python script, blindly believing what's coming in, creates an SMT component out of it, while it's a THT component.

If the base data is already wrong, I can only see 3 options:

1. * accept the rubbish data and correct it manually in Kicad
2. * try to sanitize/cross-check the Easyeda produced data, e.g. in the SMT-or-not logic, by taking also the footprint title or package into account (in this case "HDR-TH_7P-P2.54-V-F"). E.g. if it contains "-TH" while the official SMT flag being True, then make it false! Of course there's a no guarantee that this will catch all bad cases, but it's a start. Maybe it catches enough and causes no false positives...
3. * Inform Easyeda/LCSC that their libs are not correct, and maybe they should change/sanitize them. But seeing that @uPesy has removed the sponsorship banner, the cooperation might not be 100% any more. But YMMV.

[image]

Maybe there's another option, in the python code the API call specifies a version number as parameter. Maybe there's a version number providing correct data, as the Easyeda app itself also must be somehow smart enough to know what are THT and SMT parts. Or the app uses different endpoints.

Full API JSON response included for reference:
[C2932672_json.txt](https://github.com/uPesy/easyeda2kicad.py/files/11122078/C2932672_json.txt)

netweaver1970, 2023-04-01:
issue with the *.paste pads for THT fixed, full source attached. 3 code changes made. I'll try to tackle my other issues I have with imported THT 3d models later. Also, the misaligned SMT components I'll have a look be but that might be less intuitive to fix.

Just overwrite all files with the contents of this file. @uPesy maybe can do the diff and include it in the main source, so it becomes official. 
[easyeda2kicad.zip](https://github.com/uPesy/easyeda2kicad.py/files/11128842/easyeda2kicad.zip)

netweaver1970, 2023-04-04:
update, all THT components are now properly identified (so you can filter the correct ones in the 3d view) and all 'normal' THT are lowered to the base plate of the PCB. Exceptional cases, like the ESP-01 which has a double 'z' now are also glued pcb-to-pcb, which is obviously not right. But I'll try to fix that later. At least the majority of the THT's will be correct now.
2 extra (v. small) code changes made.
[image]
[easyeda2kicad.zip](https://github.com/uPesy/easyeda2kicad.py/files/11152712/easyeda2kicad.zip)

uPesy, 2023-04-05:
Thanks for your help !
I have updated files in the `dev` branch to include your modifications.
Could you verify that the new codes works on severals component ?

netweaver1970, 2023-04-06:
> Thanks for your help ! I have updated files in the `dev` branch to include your modifications. Could you verify that the new codes works on severals component ?

mmmm, I took a simple 1/8W THT resistor(C2857411) and it's not floating anymore ... but it's now put too low, split in half :)
[image]
So obviously the logic to determine the required Z translation is very different by type of object...
I feel the technical name of the package also plays a role in the resulting exact positioning/translation values, as I can't find any other field in the data that can be used to extract heights from. Unless EasyEDA is parsing the wrl file itself, the geometry. 
If that was the case, then I have little chance of replicating the same behaviour.

Maybe you still have some contacts at EasyEDA, since you were once sponsored by them, to ask them what the logic is?

For now I don't see an issue bringing it live, my changes are only for THT components, and those heights were not that great either to start with. At least the issue that the paste was also over the pads with holes, is now also fixed.

netweaver1970, 2023-04-06:
@uPesy if you want to reverse the 'pulling down' change of all THTs, then this is the only code line to revert to your code:
[image]

There all the smart(er) logic will need to be put, if we ever find out what the correct logic is.

netweaver1970, 2023-04-13:
after some extra googling, it seems that indeed the lowest point of the part geometry itself is being used in another EasyEda2KiCad component converter:
This one [https://github.com/TousstNicolas/JLC2KiCad_lib](https://github.com/TousstNicolas/JLC2KiCad_lib) uses this logic for the translate_z value:

` 
  # find the lowest point of the model
    lowest_point = 0
    for point in vertices:
        if float(point.split(" ")[2]) < lowest_point:
            lowest_point = float(point.split(" ")[2])

    # convert lowest point to mm
    min_z = lowest_point * 2.54

    # calculate the translation Z value
    from .footprint_handlers import mil2mm

    Zmm = mil2mm(translationZ)
    translation_z = Zmm - min_z
`
in the model3d.py file.

I haven't checked the quality of their logic yet on multiple objects, but it looks at least more elaborate than our crude way to determine translation_z.
Maybe also the midpoint of the model determination is better there, as in your/(tiny bit our) conversion code, the SMD's as well as THT components are not always correctly centred over the footprint as well.

But as they say, the proof of the pudding is in the eating.

uPesy, 2023-08-30:
Yes, i have trying many times to figure out the proper way of getting the correct z-index without success ...
I will check what [TousstNicolas](https://github.com/TousstNicolas/JLC2KiCad_lib) have done.

dgbalharrie, 2023-10-23:
I have seen the same problem with some through hole LCSC components imported into Kicad.

[image]

The two floating components are C206887 and C880574. The TO220 transistor is C5121605 and the diode is C2992915, both sit correctly on the board.