uPesy/easyeda2kicad.py #95: Python 3.11 Mac issue?

RoyFlashlight, 2023-07-12:
XXX@XXX-Laptop ~ % easyeda2kicad --footprint --lcsc_id=C2040      
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.11/bin/easyeda2kicad", line 5, in <module>
    from easyeda2kicad.__main__ import main
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/easyeda2kicad/__main__.py", line 12, in <module>
    from easyeda2kicad.easyeda.easyeda_importer import (
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/easyeda2kicad/easyeda/easyeda_importer.py", line 6, in <module>
    from easyeda2kicad.easyeda.parameters_easyeda import *
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/easyeda2kicad/easyeda/parameters_easyeda.py", line 26, in <module>
    class EeSymbolPinSettings(BaseModel):
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pydantic/_internal/_model_construction.py", line 131, in __new__
    cls.__pydantic_decorators__ = DecoratorInfos.build(cls)
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pydantic/_internal/_decorators.py", line 414, in build
    res.validators[var_name] = Decorator.build(
                               ^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pydantic/_internal/_decorators.py", line 249, in build
    func = shim(func)
           ^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pydantic/_internal/_decorators_v1.py", line 77, in make_generic_v1_field_validator
    raise PydanticUserError(
pydantic.errors.PydanticUserError: The `field` and `config` parameters are not available in Pydantic V2, please use the `info` parameter instead.

For further information visit https://errors.pydantic.dev/2.0.2/u/validator-field-config-info
XXX@XXX-Laptop ~ %

RoyFlashlight, 2023-07-12:
[SteamingCoax](https://github.com/SteamingCoax)

Adresses this my bad

closed 2023-07-12 by RoyFlashlight

hacktronics, 2023-08-09:
Windows also facing similar issue with Anaconda and spaces in username:
```
Traceback (most recent call last):
  File "C:\ProgramData\anaconda3\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\ProgramData\anaconda3\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\P B\AppData\Roaming\Python\Python310\Scripts\easyeda2kicad.exe\__main__.py", line 4, in <module>
  File "C:\Users\P B\AppData\Roaming\Python\Python310\site-packages\easyeda2kicad\__main__.py", line 12, in <module>
    from easyeda2kicad.easyeda.easyeda_importer import (
  File "C:\Users\P B\AppData\Roaming\Python\Python310\site-packages\easyeda2kicad\easyeda\easyeda_importer.py", line 6, in <module>
    from easyeda2kicad.easyeda.parameters_easyeda import *
  File "C:\Users\P B\AppData\Roaming\Python\Python310\site-packages\easyeda2kicad\easyeda\parameters_easyeda.py", line 26, in <module>
    class EeSymbolPinSettings(BaseModel):
  File "C:\Users\P B\AppData\Roaming\Python\Python310\site-packages\pydantic\_internal\_model_construction.py", line 130, in __new__
    cls.__pydantic_decorators__ = DecoratorInfos.build(cls)
  File "C:\Users\P B\AppData\Roaming\Python\Python310\site-packages\pydantic\_internal\_decorators.py", line 441, in build
    res.validators[var_name] = Decorator.build(
  File "C:\Users\P B\AppData\Roaming\Python\Python310\site-packages\pydantic\_internal\_decorators.py", line 249, in build
    func = shim(func)
  File "C:\Users\P B\AppData\Roaming\Python\Python310\site-packages\pydantic\_internal\_decorators_v1.py", line 77, in make_generic_v1_field_validator
    raise PydanticUserError(
pydantic.errors.PydanticUserError: The `field` and `config` parameters are not available in Pydantic V2, please use the `info` parameter instead.

For further information visit https://errors.pydantic.dev/2.1.1/u/validator-field-config-info
```