uPesy/easyeda2kicad.py #98: Shapes' `is_locked` seems to not always be boolean

ElementW, 2023-07-22:
LCSC part number `C697956` import fails with
```
pydantic.error_wrappers.ValidationError: 1 validation error for EeFootprintRectangle
is_locked
  value could not be parsed to a boolean (type=type_error.bool)
```
For some reason `is_locked` is... a float here?
```
{
    'x': '3995.5',
    'y': '2994.0005',
    'width': '6',
    'height': '10',
    'stroke_width': '3',
    'id': 'gge49',
    'layer_id': '0',
    'is_locked': '0.5'
}
```
I'm really not sure what a non-bool value could mean there, it could be a data entry error or some type of corruption on LCSC's side, only for it to get silently accepted by EasyEDA's JavaScript considering `0.5` as a truthy value.

Patching the `is_locked` type to `typing.Any` is a workaround (especially since this value doesn't' seem to be read back?) but more investigation is probably needed.

bugamehmet, 2023-07-22:
The data provided for is_locked is '0.5', which seems to be a data entry error or possibly some corruption on LCSC's side. It is likely that the correct value for is_locked should have been either '0' (False) or '1' (True) to represent the boolean values.
Keep in mind that the decision to handle is_locked as typing.Any could have implications elsewhere in your system, so it's essential to assess the impact of this change thoroughly.
The error message indicates that Pydantic was unable to parse the value of is_locked as a boolean because it received a float instead. Pydantic is strict about data types and expects the values to match the specified types.

uPesy, 2023-08-29:
@ElementW > only for it to get silently accepted by EasyEDA's JavaScript considering 0.5 as a truthy value.
I think it's will be ok since this field `is_locked ` is used by the script.
I propose to replace 
```    @field_validator("is_locked", mode="before")
    @classmethod
    def empty_str_lock(cls, field):
        return False if field == "" else field
```
by
```    @field_validator("is_locked", mode="before")
    @classmethod
    def empty_str_lock(cls, field):
        return False if field == "" else bool(float(field))
```
Could you change it also for other `BaseModel` and test it on some components to verify that is work correctly ?