uPesy/easyeda2kicad.py #101: Windows 10: files on other than C: drive?

phil-barrett, 2023-08-08:
Installed easyeda2kicad on my D: drive.  Get this message when I run it.  Is it confused about where my files are installed?  Anything I can do to resolve this?
`
D:\Documents\kicad\easyeda2kicad>python easyeda2kicad --full --lcsc_id=C2040
-- easyeda2kicad.py v0.6.3 --
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "D:\Documents\kicad\easyeda2kicad\easyeda2kicad\__main__.py", line 361, in <module>
    sys.exit(main(sys.argv[1:]))
             ^^^^^^^^^^^^^^^^^^
  File "D:\Documents\kicad\easyeda2kicad\easyeda2kicad\__main__.py", line 239, in main
    if not valid_arguments(arguments=arguments):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Documents\kicad\easyeda2kicad\easyeda2kicad\__main__.py", line 161, in valid_arguments
    os.mkdir(default_folder)
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\philba\\Documents\\Kicad\\easyeda2kicad'

D:\Documents\kicad\easyeda2kicad>`

uPesy, 2023-08-29:
I think it's related to :https://github.com/uPesy/easyeda2kicad.py/blob/92088a460fddaa6d266b28be507c863f5635e8c6/easyeda2kicad/__main__.py#L154-L159
The function https://docs.python.org/3/library/os.path.html#os.path.expanduser will create the base path based on your OS. But i don't know how to force to use a specific drive. It will use the drive when Windows is installed (C:).

uPesy, 2023-10-13:
You can maybe use a relative path instead of the default to fix this issue

tintinhamans, 2025-02-07:
The issue on Windows is that the Documents folder does not necessarily need to be located within the user profile directory (~). User is able to relocate the location of Documents to any path of their choosing.

Something like this needs to be done to resolve the current path of Documents folder correctly in cases where the path is not the default:

```python
import ctypes
from pathlib import Path

def get_documents_folder():
    CSIDL_PERSONAL = 5  # CSIDL for "My Documents"
    buf = ctypes.create_unicode_buffer(260)
    ctypes.windll.shell32.SHGetFolderPathW(None, CSIDL_PERSONAL, None, 0, buf)
    return Path(buf.value)

documents_path = get_documents_folder()
print(documents_path)

```