uPesy/easyeda2kicad.py #141: Issue Adding new part to sym library

OnlineDynamic, 2024-07-04:


channelsixlabs, 2024-07-24:
I am also experiencing this quite consistently.

zyeborm, 2024-08-21:
I get this very often as well. An example part that gave this error was C496551.
It doesn't happen every time so I wonder if it may be related to specific parts.
@OnlineDynamic's fix worked for me as well. I hadn't worked out what exactly was wrong with the files that were being made and I was up to easyeda2kicad8.kicad_sym on this project alone lol. Thanks so much!

zyeborm, 2024-08-21:
[easyeda2kicad8.txt](https://github.com/user-attachments/files/16687850/easyeda2kicad8.txt)
Figured I should put this in as an example, this is the working file after I removed the ) from before the C496551 part but hopefully it allows replication.

Luro02, 2024-08-31:
I could not reproduce this issue with the part C496551, mine was added successfully:
[image]
and the file looks good as well
[easyeda2kicad.txt](https://github.com/user-attachments/files/16824193/easyeda2kicad.txt)