uPesy/easyeda2kicad.py #142: EasyEDA 3D and schematic model problems from easyeda2kicad.py

KJ7LNW, 2024-07-13:
I am not sure if this is a problem with the original models, or if is even anything that `easyeda2kicad.py` can address them, but I thought I would make a note in case you are able to help:

Schematic problems: the connection point does not align to the Kicad schematic capture grid:
- C264536
[image]

The following part numbers have severe 3D alignment (origin-related?) issues:
- C157972
- C158012
- C2939348
- C237493
- C730069 - and this one's origin is in outer space!

[image]

- CN1,2,3 should be aligned;
-  U2,U3 are in outer space (but they do exist if you zoom out),
-  and the DIN connector should be atop of the board, not lost inside:

[image]

If you need a copy of the zip that I am using, click on the link below and enter the C number to download the same zip that I am using with the Kicad [Imparter](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin) plugin:

- https://eda.ewheeler.org/easyeda2kicad.fcgi

KJ7LNW, 2024-07-18:
We just uploaded the design to the JLC assembly website and their 3D preview looks great. The 3D view in KiCad (from easyeda2kicad) is out of place, as you can see above, but somehow JLC is using the same exact 3D models to generate the image below. It would be nice if somehow easyeda2kicad could do something to the models that are downloaded from easy EDA so that they render properly in KiCad:

[image]

KJ7LNW, 2024-07-26:
We found by selecting a footprint in a layout and pressing the keyboard shortcut `S (grid origin)` that the XYZ rotation offsets can be modified.

I'm not sure if this is generic, but we found that part# [C17513](https://eda.ewheeler.org/easyeda2kicad.fcgi?lcsc=C17513)'s XYZ positions just needed to be zero'ed.

Thus changing `-1.410` to `0.000` made KiCad render the part properly. Maybe adding new `easyeda2kicad` options like `--datum-set-xyz=0,0,0`, `--datum-set-rotation=0,0,0`, and `--datum-set-scale=1,1,1` could help solve the issue?

**original**
```scheme
(model "/tmp/Z9DWo2JMlH/part.3dshapes/R0805_L2.0-W1.3-H0.6.wrl"
        (offset (xyz -1.410 -0.000 -0.000))
        (scale (xyz 1 1 1))
        (rotate (xyz 0 0 0))
    )
```

**fixed**
```scheme
(model "/tmp/Z9DWo2JMlH/part.3dshapes/R0805_L2.0-W1.3-H0.6.wrl"
        (offset (xyz 0.000 0.000 0.000))
        (scale (xyz 1 1 1))
        (rotate (xyz 0 0 0))
    )
```

Luro02, 2024-09-18:
I have the same issue, but with different components, so I will post them here, in case that helps narrowing down the issue:
- IND-SMD_L5.0-W5.0 (C7427132)
[image]
- SOD-523_L1.2-W0.8-LS1.6-Bl (C20416613)
[image]
- SOT-23-3_L2.9-W1.3-P1.90-LS2.4-BR (C489349)
[image]
- SOT-23-6_L2.9-W1.6-P0.95-LS2.8-BL (C191884)
[image]
- SW-SMD_DSHP03TS-S (C319051)
[image]
- SW-SMD_TS-1086E-AC03526 (C720473)
[image]
- USB-C_SMD-TYPE-C-31-M-12 (C165948)
[image]