uPesy/easyeda2kicad.py #169: Tool stops working if you modify a symbol

shaun-wild, 2025-06-13:
I'm not sure if I'm using the tool incorrectly here, but if I import a symbol into KiCad, modify said symbol (For example to consolidate pins, and specify outputs etc), then any subsequent imports from the tool will not add anything to my library.

Abillow1, 2025-06-14:
Oh that might be whats happening to me, Im having the same issue. The new things are going into the file but they arent appearing into Kicad, but all of the older ones stay.

shaun-wild, 2025-06-14:
@wolf-yuan-6115 This resolved the issue for me originally, but now my file seems to be corrupted beyond repair, I have even validated it with an s expression validator.

wolf-yuan-6115, 2025-06-14:
> [@wolf-yuan-6115](https://github.com/wolf-yuan-6115) This resolved the issue for me originally, but now my file seems to be corrupted beyond repair, I have even validated it with an s expression validator.
> 
> Seems like a gross oversight from this tool - I don't think inserting text based on regular expressions is the best solution, a more robust solution could be to use a proper S-Expression parser.

That wasn't in my expectation, I only repair this once and haven't add new symbol into the file afterwards, probably needs original author to look into this.

wolf-yuan-6115, 2025-09-01:
Perhaps https://github.com/uPesy/easyeda2kicad.py/pull/166 will fixed this?